lvgl/lvgl_pro #5: Feedback on UI editor

roblatour, 2025-01-04:
I've just seen the youtube video on the development of UI editor and I would like sharing my feedback.

Just wondering if you have considered building this as a Visual Studio Code extension instead?

Might save you a lot of work around the editor and its features and let you focus on the core of what you are trying to achieve.

https://code.visualstudio.com/docs/languages/markdown as an example; it lets you code in one pain and see the results in another.

Maybe this is not at all possible for what you want to do, but if it is it may very well be a good alternative.  

Additionally, it would reduce the learning curve for people + make adoption easier.

kisvegabor, 2025-01-15:
Hey,

Making the  editor availalble as a VSCode extension is absolutely on the roadmap. Actually it's planned for v0.2. :slightly_smiling_face:

apavelm, 2025-08-09:
@mrhubbs You're right, it helped! At least for the "examples" project. Thanks a lot.
It is not something straight forward and confusing. Looking forward to a fix.

BTW, Apart of that, I'd suggest creating a feature to reverse compilation, like you have a c/c++ function with the creation of LVGL elements and according to that XML file composing.

closed 2025-11-08 by mrhubbs