lvgl/lvgl_pro #6: Error when saving file

makerinchina-iot, 2025-01-12:
First of all, thank you very much for providing this tool. I’m very pleased to see the first version come out so quickly, and I couldn’t wait to try it out. But I’ve encountered a minor issue here and need your help.

When I save the file after editing it, the error shows in the console:

```
Error saving file: Error: Error invoking remote method 'save-file': Error: ENOENT: no such file or directory, open 'D:\d%3A\lvgl_editor\examples\components\home.xml'
```

The example file is in the directory D:\lvgl_editor\examples, and I use Win11 OS.

ptu14, 2025-01-12:
It looks like the issue is caused by the file path being URL-encoded (`d%3A` instead of `d:`) and not properly decoded before the file operation.

coevin, 2025-01-14:
I can confirm that this is indeed a bug on Windows. 

There appear to be two issues. As mentioned by @ptu14 part of the path is being URL-encoded. In my case c%3A shold be c:. But there is already a drive letter prefixed as well. So even if it were decoded in place it would still cause an issue. 

`Error saving file: Error: Error invoking remote method 'save-file': Error: ENOENT: no such file or directory, open 'C:\c%3A\Users\user\Documents\lvgl_editor\examples\components\home.xml'`

coevin, 2025-01-14:
@makerinchina-iot 

I have found and fixed the issue. Please use this link for a new installer: https://drive.google.com/file/d/1jG1kqLm3mkylcwYAQDkcPDySbCLBqxhW/view?usp=sharing

makerinchina-iot, 2025-01-14:
> I've also updated [the installer](https://github.com/lvgl/lvgl_editor/releases/tag/v0.1.0). (I didn't want to create a new release for it)

Thank you very much for your reply. I reinstalled it and it's working properly now.

closed 2025-01-18 by makerinchina-iot