lvgl/lvgl_pro #8: It doesn't do anything.

kdschlosser, 2025-01-13:
I installed it (Windows 11) and when it runs it prompts me to open a project. Now sure how to open a project when I have not been given the chance to create one. So I click the button. It then asks me to select a directory and I do. Then it drops me back to the UI editor where it states that I have not opened a project and provides a button to select a directory.  There are a couple of other buttons but nothing that is relevent to creating a new project. There are buttons for "Compile Project", "Generate Code" and "Clear". If I click on "Generate code I get this as output in the console window

```
Console Ready
Generating code
Generating code for project: ****************
Failed to generate code.
Code generated successfully
```

The last 2 lines are rather interesting because is says it failed and then it says it was successful. 

Pressing the "Compile" button gives this output. 

```
Copying to temp directory.
liblvgl.a not found, starting build process...
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
Command failed: Error: connect ENOENT //./pipe/docker_engine
Failed to build LVGL library: Error: Docker command failed: Error: connect ENOENT //./pipe/docker_engine
Build failed: Failed to build liblvgl.a
```

Which it takes quite a while to get past the Copying to temp directory. Not sure why as it is running on a 32 logical processor machine with 128mb of memory with an SSD as the drive. It should have taken a fraction of a second to do the copy of a folder that has no files in it.

kdschlosser, 2025-01-13:
Suggestion to change the word "Console" to " Output Log". Console implies the user has the ability to directly interact with it and that is not the case. It is only to display an output log.

kisvegabor, 2025-01-13:
Hi,

Please check [this video](https://youtu.be/gCxBAK9EByA) and follow the instructions in it. There you will see that you should clone this repo and use it as an example.

kdschlosser, 2025-01-13:
why not package the example into the program installer and have the open directory default to that location the first time it is opened?

kdschlosser, 2025-01-13:
well at least it now shows the files but it still doesn't work properly.
```
Console Ready
Generating code
Generating code for project: C:\Users\drsch\Documents\lvgl_editor\examples
Code generated successfully
Compiling project
Copying to temp directory.
liblvgl.a not found, starting build process...
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
Command failed: Error: connect ENOENT //./pipe/docker_engine
Failed to build LVGL library: Error: Docker command failed: Error: connect ENOENT //./pipe/docker_engine
Build failed: Failed to build liblvgl.a
```

I am not sure as to why it is making a temporary directory in the root of my HDD instead of using the system defined temporary directory which is located in `%APPDATA%\Local\Temp`. The funny thing is there is no `tmp` directory that gets created in the root of my drive and there isn't one that gets created in the system supplied temp folder either. I am going to pin a debugger to it to see what the app is opening for handles and where those handles point to.

MakeFile is only available on Posix systems or inside of virtual machines like MinGW. It is not something that is available on Windows. CMake is what gets used to generate the files needed by Ninja. Ninja is what gets used as the build system when compiling on Windows.

kdschlosser, 2025-01-13:
https://forums.docker.com/t/docker-on-windows-fails-with-pipe-docker-engine-the-system-cannot-find-the-file-specified/28479/3

kisvegabor, 2025-01-13:
@coevin please take a look at this issue.

coevin, 2025-01-13:
Docker needs to be running

kisvegabor, 2025-01-13:
Yes, Docker is required now. Later it will be needed only for advanced stuff such as rebuilding the preview of the editor.

kdschlosser, 2025-01-13:
why is docker a requirement at all?

kdschlosser, 2025-01-13:
why would the editor need to be recompiled?

kisvegabor, 2025-01-14:
Please watch the video.

kdschlosser, 2025-01-14:
You can explain it all you like in a video but it doesn't do me any good. I don't have sound on my computer so all I would be seeing is your lips flapping away and me having no clue what you are saying... LOL

juneofive, 2025-02-24:
for me too,the first issue (the biggest one) is installing Docker, although for some people this might not be a problem.

kisvegabor, 2025-02-24:
I hear you. We are already discussing how to remove Docker and make the Editor easier to get started with.