lvgl/lvgl_pro #9: Compile problem

caballero03, 2025-01-14:
Hello,

I have followed the instructions given precisely however, I am getting an error when trying to compile the project. I don't know what I need to change to make this work.

Running Ubuntu 20.04 with Docker installed and working. I did fork the repo and enabled the actions and pages function as described in the instructions.  

Please advise as I am hoping to try out the editor and give some feedback. Thank you.

Here is the contents of my console window post compile:

```
Console Ready
Compiling project
Copying to temp directory.
liblvgl.a not found, starting build process...
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
�configure: cmake /work/lib -DCMAKE_TOOLCHAIN_FILE=/emsdk/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake -DCMAKE_CROSSCOMPILING_EMULATOR=/emsdk/node/20.18.0_64bit/bin/node-- Configuring done
-- Generating done
0-- Build files have been written to: /tmp/build
make: make -j4
.make: /usr/bin/cmake: Operation not permitted
>make: *** [Makefile:9953: cmake_check_build_system] Error 127
.emmake: error: 'make -j4' failed (returned 2)

Failed to build LVGL library: Error: liblvgl.a was not created
Build failed: Failed to build liblvgl.a
```

kisvegabor, 2025-01-14:
Please share the result of `docker --version`.

caballero03, 2025-01-14:
```
~/Documents/projects/src$ docker --version
Docker version 20.10.9, build c2ea9bc
```

kisvegabor, 2025-01-15:
Mine is 
```
Docker version 27.1.1, build 6312585
```

I don't know if it could be related, but good to keep in mind for @coevin.

coevin, 2025-01-15:
@caballero03 @kisvegabor 

I will try to spin up a Ubuntu 20.04 with Docker instance to see if I can duplicate, and report back.

coevin, 2025-01-17:
@caballero03 

I was not able to duplicate using Docker 27.5.0 on a fresh install of Ubuntu 20.04. I can't necessarily confirm that Docker is the issue, but I can try and see if I can install 20.10.09 to confirm.

FikaBaby, 2025-02-21:
Not finding ui.h ... but dont really know what I am doing... Be grateful for help!?

Console Ready
Compiling project
Copying to temp directory.
liblvgl.a already exists
Created clean build directory
Preparing CMake process
Running CMake process
CMake process completed
Preparing make process
Running make process
Make failed:
make: make -j8
U[ 33%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_default_gen.c.o
N[ 33%] Building C object CMakeFiles/lved-runtime.dir/src/getrandom_compat.c.o
S[ 50%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_error_gen.c.o
U[ 66%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_warning_gen.c.o
E[ 83%] Building C object CMakeFiles/lved-runtime.dir/src/runtime.c.o
I/user_src/button_default_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
I/user_src/button_warning_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
1 error generated.
1 error generated.
~make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:77: CMakeFiles/lved-runtime.dir/user_src/button_default_gen.c.o] Error 1
-make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:107: CMakeFiles/lved-runtime.dir/user_src/button_warning_gen.c.o] Error 1
G/user_src/button_error_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
1 error generated.
|make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:92: CMakeFiles/lved-runtime.dir/user_src/button_error_gen.c.o] Error 1
=/work/src/runtime.c:2:10: fatal error: 'ui.h' file not found
    2 | #include "ui.h"
      |          ^~~~~~
1 error generated.
omake[2]: *** [CMakeFiles/lved-runtime.dir/build.make:137: CMakeFiles/lved-runtime.dir/src/runtime.c.o] Error 1
Pmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lved-runtime.dir/all] Error 2
%make: *** [Makefile:91: all] Error 2
.emmake: error: 'make -j8' failed (returned 2)

Build failed: Make failed:
make: make -j8
U[ 33%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_default_gen.c.o
N[ 33%] Building C object CMakeFiles/lved-runtime.dir/src/getrandom_compat.c.o
S[ 50%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_error_gen.c.o
U[ 66%] Building C object CMakeFiles/lved-runtime.dir/user_src/button_warning_gen.c.o
E[ 83%] Building C object CMakeFiles/lved-runtime.dir/src/runtime.c.o
I/user_src/button_default_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
I/user_src/button_warning_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
1 error generated.
1 error generated.
~make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:77: CMakeFiles/lved-runtime.dir/user_src/button_default_gen.c.o] Error 1
-make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:107: CMakeFiles/lved-runtime.dir/user_src/button_warning_gen.c.o] Error 1
G/user_src/button_error_gen.c:12:10: fatal error: 'ui.h' file not found
   12 | #include "ui.h"
      |          ^~~~~~
1 error generated.
|make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:92: CMakeFiles/lved-runtime.dir/user_src/button_error_gen.c.o] Error 1
=/work/src/runtime.c:2:10: fatal error: 'ui.h' file not found
    2 | #include "ui.h"
      |          ^~~~~~
1 error generated.
omake[2]: *** [CMakeFiles/lved-runtime.dir/build.make:137: CMakeFiles/lved-runtime.dir/src/runtime.c.o] Error 1
Pmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lved-runtime.dir/all] Error 2
%make: *** [Makefile:91: all] Error 2
.emmake: error: 'make -j8' failed (returned 2)

coevin, 2025-02-21:
> Not finding ui.h ... but dont really know what I am doing... Be grateful for help!? UBUNTU 20.0.4.

Hi @FikaBaby , can you create a fresh new issue for this, as it doesn't seem to relate to the original issue reported here.

jaychub, 2025-03-16:
same error as above using windows 10 pro and Docker version 20.10.2, build 2291f61

Console Ready
Compiling project
Copying to temp directory.
liblvgl.a not found, starting build process...
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
Command failed: Error: connect ENOENT //./pipe/docker_engine

coevin, 2025-03-16:
@jaychub `Command failed: Error: connect ENOENT //./pipe/docker_engine` I think usually implies that Docker isn't running. Could you please confirm whether that is the case or not?

jaychub, 2025-03-16:
updated docker to the latest version : docker --version
Docker version 27.5.1, build 9f9e405

and tried to compile and i get this error:
Console Ready
Compiling project
Copying to temp directory.
liblvgl.a not found, starting build process...
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
Command failed: Error: (HTTP code 404) no such container - No such image: coevin/emscripten-sdl2:main 
Failed to build LVGL library: Error: Docker command failed: Error: (HTTP code 404) no such container - No such image: coevin/emscripten-sdl2:main 
Build failed: Failed to build liblvgl.a

jaychub, 2025-03-16:
here is how i fixed it:
i switched to using WSL2 as my backend instead of windows. and here is how:
make sure wsl is installed:
1. Install WSL 2 (if not installed)
Open PowerShell as Administrator and run:
sh
Copy
Edit
wsl --install
Restart your computer.
Set WSL 2 as the default backend:
sh
Copy
Edit
wsl --set-default-version 2

then 
2. Enable Docker WSL 2 Backend

[image]

Open Docker Desktop.
Go to Settings > General.
Enable "Use the WSL 2 based engine".
Apply and restart Docker

then Switch Docker to Linux Containers:

[image]

Right-click on the Docker icon in the system tray.
Select "Switch to Linux Containers".
Wait for Docker to restart.

and finally restart the editor, and when that happened it started downloading the docker image :

[image]

and finally it worked :)

[image]

coevin, 2025-03-16:
@jaychub that's a lot of hoops to have to jump through. I was just reading something similar on [Stack Overflow](https://stackoverflow.com/questions/50786984/docker-image-operating-system-linux-cannot-be-used-on-this-platform-while)

jaychub, 2025-03-16:
it really boils down to switching Docker to use Linux as it's backend and using WSL2 instead of hyper-v as the engine. so now it works maybe this could be added in your readme for the windows users.

coevin, 2025-03-17:
> maybe this could be added in your readme for the windows users

@jaychub yes, fully agree

gooddaddy, 2025-04-25:
Console Ready
Compiling project
⏳ Compiling project: /home/liyubo/Documents/workspace/lvgl_editor/examples
❌ lib LVGL build failed: Error: (HTTP code 404) no such container - No such image: lvglio/emscripten-sdl2:0.1.0 
Build failed: Error: Failed to build liblvgl.a
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module

kisvegabor, 2025-04-25:
Hey,
Please start the editor in a terminal and copy the full log from there.

Please also check if docker is stalled and available correctly. Just type `docker --version` in a terminal.

Also:
- Which OS do you use?
- Do you use the old v0.1 or the recent v0.2-rc version? 

cc @mrhubbs

closed 2025-07-23 by mrhubbs