lvgl/lvgl_pro #12: Compile Problem win11

jwillner, 2025-01-16:
Configuring LVGL library build...
Running command: cd /work/lib &&       mkdir -p /tmp/build && cd /tmp/build &&       emcmake cmake /work/lib &&       emmake make -j$(nproc) &&       cp liblvgl.a /work/lib/
Command failed: Error: connect ENOENT //./pipe/docker_engine
Failed to build LVGL library: Error: Docker command failed: Error: connect ENOENT //./pipe/docker_engine
Build failed: Failed to build liblvgl.a

coevin, 2025-01-16:
@jwillner That error usually indicates Docker isn't running. Docker is a requirement.

jwillner, 2025-01-16:
there is this error shown:

```
jw@jw-virtual-machine:~$ docker info
Client: Docker Engine - Community
 Version:    27.5.0
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.19.3
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.32.4
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
ERROR: permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.47/info": dial unix /var/run/docker.sock: connect: permission denied
errors pretty printing info
jw@jw-virtual-machine:~$
```

jwillner, 2025-01-16:
I have found this, and now it is working:

[image]

kisvegabor, 2025-01-16:
Awesome! I've already checked some Docker alternatives. `podman` seems much simpler to install.

closed 2025-07-23 by mrhubbs