lvgl/lvgl_pro #13: LVGL editor download shows virus and download failed

namanjain7, 2025-02-01:
Hello,
I am trying to download the lvgl editor app for windows.
[LVGL_Editor-Windows-0.1.0.zip](https://github.com/lvgl/lvgl_editor/releases/download/v0.1.0/LVGL_Editor-Windows-0.1.0.zip)

This download is getting failed due to chrome and other antivirus detecting virus. Is this false positive case? Just wanted to confirm this.

namanjain7, 2025-02-01:
[image]

mlittle408, 2025-02-01:
[image]

coevin, 2025-02-03:
@namanjain7 

I could reproduce. But it seems like a false positive.

I did some further investigation, and this seems to have affected other Electron based programs in the past. Particularly those with an NSIS installer. Having an .msi installer instead may solve the problem. 

So we will see about producing one, and making it available for download after testing it for any virus/trojan.

[image]