lvgl/lvgl_pro #14: question: Embedded build working already?

mkiepfer, 2025-02-09:
I am currently trying to evaluate usage of the UI-Editor for an embedded build. Unfortunately this seems to be quite difficult. I am aware that there are restrictions for now. We mainly try to evaluate the workflow of UI-Editor and how we could manage to integrate and encapsulate screens and components.

Has someone already succeeded in using the UI-Editor for embedded builds?

mkiepfer, 2025-02-10:
I generally managed to get the sample project running on an embedded device (esp32). Unfortunalty the workflow is far from usable:
* The biggest issue is the proper linkage of resources (images, fonts)
* I couldn't find a way to bind events to UI elements. I am also unsure about the intended procedure
* Code-Generation needs better more modularity. E.g. every component should directly include all includes for the components it depends on

Of course this is the first pre-release. Maybe this information is helpful to prioritze the following features. I am still willing to stay an early adopter. The overall apraoch is still convincing!

kisvegabor, 2025-02-24:
> Has someone already succeeded in using the UI-Editor for embedded builds?

I don't think so, and it wasn't the goal. v0.2, released at the end of March should be usable editor. v0.3 (end of June) should be a feature complete Editor, and v1.0 (end of September) should be a production ready version. So something more useful comes soon 😊 
 
> Of course this is the first pre-release. Maybe this information is helpful to prioritze the following features. I am still willing to stay an early adopter. The overall apraoch is still convincing!

It's really useful! v0.2 will
- support images and fonts as C array
- events
- better include hierarchy.

jeffwelder-ellenbytech, 2025-05-08:
Conjured some AI slop to address the issues I have to include the generated files into my builds. :)

1) Rename #include "lvgl/lvgl.h" to #include "lvgl.h" (These should respect LV_LVGL_H_INCLUDE_SIMPLE) 
2) Replace static images with direct linkage via `LV_IMG_DECLARE(yyy_data);` &&
`lv_image_set_src(lv_image_x, &yyy_data);`

`lvgl_fixups.py`
```
import os
import sys
import shutil
import re

def replace_include_in_file(file_path):
    """Replace the specified include statement in a single .h file."""
    # Create a backup of the original file
    backup_path = file_path + ".bak"
    shutil.copy2(file_path, backup_path)
    
    try:
        # Read file content
        with open(file_path, 'r', encoding='utf-8') as file:
            content = file.read()
        
        # Replace the include statement
        modified_content = content.replace('#include "lvgl/lvgl.h"', '#include "lvgl.h"')
        
        # Write modified content back if changes were made
        if content != modified_content:
            with open(file_path, 'w', encoding='utf-8') as file:
                file.write(modified_content)
            print(f"Updated include in: {file_path}")
            return True
        else:
            # Remove backup if no changes
            os.remove(backup_path)
            return False
            
    except Exception as e:
        print(f"Error processing {file_path}: {e}")
        # Restore from backup
        shutil.copy2(backup_path, file_path)
        return False

def update_lv_image_calls(file_path):
    """Update lv_image_set_src() calls in a .c file."""
    # Create a backup of the original file
    backup_path = file_path + ".bak"
    shutil.copy2(file_path, backup_path)
    
    try:
        # Read file content
        with open(file_path, 'r', encoding='utf-8') as file:
            content = file.read()
        
        # Use a regex pattern to find lv_image_set_src calls
        pattern = r'(lv_image_set_src\s*\(\s*)([^,]+)(\s*,\s*)([^);\s]+)(\s*\))'
        
        # Function to process each match and create replacement
        def replacement(match):
            full_match = match.group(0)
            prefix = match.group(1)  # lv_image_set_src(
            first_param = match.group(2)  # first parameter
            comma = match.group(3)  # ,
            second_param = match.group(4).strip()  # second parameter (cleaned)
            suffix = match.group(5)  # )
            
            # Remove any whitespace or & if present in second_param
            clean_second_param = second_param.strip('& \t\n\r')
            
            # Create the declaration and updated function call
            declaration = f"LV_IMG_DECLARE({clean_second_param}_data);\n"
            new_call = f"{prefix}{first_param}{comma}&{clean_second_param}_data{suffix}"
            
            return declaration + new_call
        
        # Apply the replacements
        modified_content = re.sub(pattern, replacement, content)
        
        # Write modified content back if changes were made
        if content != modified_content:
            with open(file_path, 'w', encoding='utf-8') as file:
                file.write(modified_content)
            print(f"Updated lv_image_set_src calls in: {file_path}")
            return True
        else:
            # Remove backup if no changes
            os.remove(backup_path)
            return False
            
    except Exception as e:
        print(f"Error processing {file_path}: {e}")
        # Restore from backup
        shutil.copy2(backup_path, file_path)
        return False

def process_directory(directory):
    """Process all .h and .c files in the given directory and its subdirectories."""
    h_files_processed = 0
    h_files_updated = 0
    c_files_processed = 0
    c_files_updated = 0
    
    for root, _, files in os.walk(directory):
        for filename in files:
            file_path = os.path.join(root, filename)
            
            if filename.endswith('.h'):
                h_files_processed += 1
                if replace_include_in_file(file_path):
                    h_files_updated += 1
                    
            elif filename.endswith('.c'):
                c_files_processed += 1
                if update_lv_image_calls(file_path):
                    c_files_updated += 1
    
    return {
        "h_files_processed": h_files_processed,
        "h_files_updated": h_files_updated,
        "c_files_processed": c_files_processed,
        "c_files_updated": c_files_updated
    }

if __name__ == "__main__":
    if len(sys.argv) != 2:
        print("Usage: python lvgl_fixups.py <directory>")
        sys.exit(1)
    
    target_dir = sys.argv[1]
    if not os.path.isdir(target_dir):
        print(f"Error: {target_dir} is not a valid directory")
        sys.exit(1)
    
    stats = process_directory(target_dir)
    print("\nProcessing completed:")
    print(f"- Examined {stats['h_files_processed']} .h files, updated {stats['h_files_updated']} files.")
    print(f"- Examined {stats['c_files_processed']} .c files, updated {stats['c_files_updated']} files.")
```

mrhubbs, 2025-07-21:
@mkiepfer Please try in v0.3.0, just released. If there is no CMakeLists.txt in a project, the Editor will generate one which can be extended to support embedded builds.