lvgl/lvgl_pro #15: bug: Wrong XML file overwritten with globals.xml content

mkiepfer, 2025-02-10:
It appeared several times already, that a component xml (e.g. home.xml) has been overwritten with the content of the `globals.xml` file content. I am unsure about reproducabiility, but I guess this happens, when the generate or the project build button have been triggered.

This is quite annoying and blocks usability. Sometimes the file needs to be reopened before the change is visable, as the UI-Editor seems to be not able to detect file system changes.

mkiepfer, 2025-02-10:
This issue seems to only appear on MacOS.

mrhubbs, 2025-02-17:
@mkiepfer Thank you for the feedback. Refreshing the UI-Editor content's when a file is changed externally is an enhancement we have planned. Overwriting an XML with `globals.xml` is not something I've seen. I'll try to reproduce.

closed 2025-07-30 by mrhubbs