lvgl/lvgl_pro #16: adjusting width and height of virtual preview device

minovap, 2025-02-12:
is it possible to change the width and height of the preview? i cant seem to find where you set the width and height in the project.  i would like to try building pages for an embedded device in its native resolution

mrhubbs, 2025-02-17:
Hi @minovap, this is a feature we are working on.

FikaBaby, 2025-02-21:
I want to use a Arduino Giga display, 480 x 800, what do we have here? Still an awesome learning platform for LVGL! Just got it going and learning Linux.

h3lpd3sk, 2025-02-23:
Please include round and rounded displays as well

kisvegabor, 2025-02-24:
We add a `<preview>` tag to components in v0.2 as described [here](https://docs.lvgl.io/master/details/other-components/xml/preview.html). By using
```xml
    <preview name="round"  width="400" height="400" style_radius="30" style_clip_corner="true"/>
```
you should be able to get a rounded screen. 

We will make sure that this scenario works well. 

cc @richardgazdik

kisvegabor, 2025-10-20:
`<preview>` is already implemented, so I close this issues.

closed 2025-10-20 by kisvegabor