lvgl/lvgl_pro #20: Widget XML overwritten with C code

jaychub, 2025-03-17:
Hello,  

While following the video, I updated the color variable to RED as shown. After hitting compile, an error appeared, and the preview stopped working.  

Although slider_box.xml appears correctly in the editor, the actual slider_box.xml file contains C code instead of XML when opened in Notepad++. The same issue occurs with dark_slider.xml—both files seem to be overwritten with C code.  

Reopening the editor loads the XML files exactly as they appear in Notepad++, with C code instead of XML. Additionally, closing and reopening slider_box.xml within the editor does not reload the file from disk. Instead, it seems to reload from memory or a temporary location. When I replaced the file manually, the editor did not reflect the changes correctly, and I had to close and reopen the editor to see the actual file.  

I also noticed that when I hit **Ctrl + S** to save changes, slider_box.xml gets replaced with C code from dark_slider.c, while dark_slider.c remains unchanged. It seems like the wrong file is being overridden, and this issue occurs when multiple files are open in the editor.  

Any idea what might be causing this?

jaychub, 2025-03-17:
Update doing the changes outside of the editor like in VSCODE and coming back to the editor and hit compile works and the changes are visible without errors, doing the changes inside the editor messes things up. the compiler throws errors. so something is definitely wrong with the editor.

kisvegabor, 2025-03-25:
Thank for reporting it. We are tracking it internally.

mrhubbs, 2025-07-21:
Fixed around or before v0.3.0