lvgl/lvgl_pro #24: callback prototype not generated

MrMarteng, 2025-04-22:
When adding a callback to a view the according prototype is not automatically generated, causing a compile error.

´´´
 <view extends="lv_obj" styles="main">
        <lv_button>
           <lv_event-call_function callback="evtButton1Click" trigger="clicked" />
        </lv_button>
</view>
´´´

This will generate in the `...gen.c` file.
´´´
lv_obj_add_event_cb(lv_button_1, evtButton1Click, LV_EVENT_CLICKED, NULL);
´´´

But the according prototype in the `...gen.h` is missing.

This has been tested with version v0.2-rc1.

kisvegabor, 2025-04-23:
Hey,

Thank you for testing it! 🚀 

I took me some time to figure out how to approach it, but finally I think I have a good idea:

- In `example_gen.h` collect all callbacks from `<lv_event-call_function>`s  and export a prototypes
- In `example_gen.c` export function definitions like this 
```c
#if LV_EDITOR_PREVIEW
void  __attribute__((weak)) callback1(lv_event_t * e)
{
   LV_UNUSED(e);
   LV_LOG("callback1 was called\n");
}
...
#endif
```
  - Add `LV_EDITOR_PREVIEW` as a compiler flag. 
```

This way we can always compile the code in the editor. What do you think?

mrhubbs, 2025-07-26:
Fixed prior to v0.3.0

closed 2025-07-26 by mrhubbs