lvgl/lvgl_pro #27: Compiling Error v0.2

redjr16, 2025-04-24:
alert: I'm a noob.  Trying to follow along with the YouTube video.  I thought I had all the components install correctly - but maybe not.  After I click the compile button, it starts with the compile messages and then displays this message;

[image]

It appears it generates the code properly.  It appears this may be a Docker related error.  When I click on button_default.xml the XML code is displayed in the windows, but there's no graphic button shown in the preview pane.

Thx,
Rick D

mrhubbs, 2025-04-24:
That error normally means Docker is not running. However, we have had some issues with Docker on WSL.

mrhubbs, 2025-04-24:
We had a user report on Docker WSL issue here: https://github.com/lvgl/lvgl_editor/issues/12

And perhaps this fix/workaround would work for you? https://github.com/lvgl/lvgl_editor/issues/9#issuecomment-2727722390

kisvegabor, 2025-10-20:
Since we don't use Docker anymore (using [Podman instead](https://docs.lvgl.io/master/details/xml/editor/install.html)), I close this issue.