lvgl/lvgl_pro #28: Compile Error - !sh: 1: cd: can't cd to /work/lib

adrenalingod, 2025-04-25:
Hay guys,

i am trying to follow the tutorial to get the LVGL UI Editor up and running on my Ubuntu LapTop with CommandLine Docker:
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
Docker version 28.1.1, build 4eba377

But I am getting an error compiling the project while every step I followed seemed to succeed.
I am getting following error:

`
Compiling project
⏳ Compiling project: /home/saruman/LVGL_Editor/lvgl_editor/examples
⏳ Building lib LVGL
!sh: 1: cd: can't cd to /work/lib

❌ lib LVGL build failed: Error: ❌ Failed to build lilb LVGL
Build failed: Error: Failed to build liblvgl.a
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
`

I tried the same with my windows 11 os with Docker Desktop.

`
Console Ready
Downloading (6%)Compiling project
Downloading (6%)⏳ Compiling project: /lvgl_editor/examples
❌ lib LVGL build failed: Error: (HTTP code 404) no such container - No such image: lvglio/emscripten-sdl2:0.1.0 
Build failed: Error: Failed to build liblvgl.a
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
Downloading (11%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Extracting (8%)
Downloading (8%)Pull complete (8%)
Setting up. Please wait a moment...
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
Extracting (0%)
Pull complete (0%)
{"status":"Digest: sha256:95abf9c65f20c600f48808a41ff017872bab82ac922461da1b05014"}
{"status":"Status: Downloaded newer image for lvglio/emscripten-sdl2:0.1.0"}
Docker image is ready
Compiling project
⏳ Compiling project:/lvgl_editor/examples
⏳ Building lib LVGL
!sh: 1: cd: can't cd to /work/lib

❌ lib LVGL build failed: Error: ❌ Failed to build lilb LVGL
Build failed: Error: Failed to build liblvgl.a
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
`

My guess is that something is wrong with the docker step of the installation.
Maybe I am forgetting something obvious what is not mentioned in the tutorial.
Any ideas?

Thanks in advance!

mrhubbs, 2025-04-27:
It certainly looks Docker-related. Can you verify your machine has the image `lvglio/emscripten-sdl2:0.1.0`? Then, on opening a project, first generate code then try compiling.

adrenalingod, 2025-04-27:
[image]

[image]

LuisMperezM, 2025-04-27:
I have the same error, and the first compilation did work and I reloaded it and the same thing happened, were you able to fix it?

adrenalingod, 2025-04-27:
No. I have no idea what to do.
My options are SquareLine or without editor

adrenalingod, 2025-04-27:
Ok. In the video and the readme it was not clear that you have to "Generate Code" first and then "Compile" now evrything works.

closed 2025-04-27 by adrenalingod

mrhubbs, 2025-04-28:
Yes, it is not clear right now. We will be updating docs and putting out a new tutorial video when this pre-release is released. :)

moonlight729, 2025-07-07:
I got the same error.
The officier did not post a clear doc to tell people how to create a environmnent of UI Editor.