lvgl/lvgl_pro #29: Have errors

LuisMperezM, 2025-04-26:
When compiling I get all that, I'm doing it from Windows

Ready
Compiling project
Copying to temp directory.
liblvgl.a already exists
Created clean build directory
Preparing CMake process
Running CMake process
CMake process completed
Preparing make process
Running make process
Make failed:
make: make -j8
K[  5%] Building C object CMakeFiles/lved-runtime.dir/user_src/examples.c.o
O[ 15%] Building C object CMakeFiles/lved-runtime.dir/user_src/examples_gen.c.o
E[ 15%] Building C object CMakeFiles/lved-runtime.dir/user_src/ui.c.o
N[ 26%] Building C object CMakeFiles/lved-runtime.dir/src/getrandom_compat.c.o
E[ 26%] Building C object CMakeFiles/lved-runtime.dir/src/runtime.c.o
f[ 31%] Building C object CMakeFiles/lved-runtime.dir/user_src/widgets/dark_slider/dark_slider_gen.c.o
b[ 36%] Building C object CMakeFiles/lved-runtime.dir/user_src/widgets/dark_slider/dark_slider.c.o
m[ 42%] Building C object CMakeFiles/lved-runtime.dir/user_src/widgets/dark_slider/dark_slider_xml_parser.c.o
`[ 47%] Building C object CMakeFiles/lved-runtime.dir/user_src/widgets/slider_box/slider_box.c.o
d[ 52%] Building C object CMakeFiles/lved-runtime.dir/user_src/widgets/slider_box/slider_box_gen.c.o
_/user_src/examples_gen.c:81:48: error: too many arguments to function call, expected 2, have 3
B   81 |         lv_xml_register_font(NULL, "inter_sm", inter_sm);
@      |         ~~~~~~~~~~~~~~~~~~~~                   ^~~~~~~~
U/work/lvgl/src/others/xml/lv_xml.h:38:13: note: 'lv_xml_register_font' declared here
U   38 | lv_result_t lv_xml_register_font(const char * name, const lv_font_t * font);
S      |             ^                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
_/user_src/examples_gen.c:82:48: error: too many arguments to function call, expected 2, have 3
B   82 |         lv_xml_register_font(NULL, "inter_md", inter_md);
@      |         ~~~~~~~~~~~~~~~~~~~~                   ^~~~~~~~
U/work/lvgl/src/others/xml/lv_xml.h:38:13: note: 'lv_xml_register_font' declared here
U   38 | lv_result_t lv_xml_register_font(const char * name, const lv_font_t * font);
S      |             ^                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
_/user_src/examples_gen.c:83:48: error: too many arguments to function call, expected 2, have 3
B   83 |         lv_xml_register_font(NULL, "inter_xl", inter_xl);
@      |         ~~~~~~~~~~~~~~~~~~~~                   ^~~~~~~~
U/work/lvgl/src/others/xml/lv_xml.h:38:13: note: 'lv_xml_register_font' declared here
U   38 | lv_result_t lv_xml_register_font(const char * name, const lv_font_t * font);
S      |             ^                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
_/user_src/examples_gen.c:85:50: error: too many arguments to function call, expected 2, have 3
E   85 |         lv_xml_register_image(NULL, "lvgl_logo", lvgl_logo);
C      |         ~~~~~~~~~~~~~~~~~~~~~                    ^~~~~~~~~
V/work/lvgl/src/others/xml/lv_xml.h:42:13: note: 'lv_xml_register_image' declared here
P   42 | lv_result_t lv_xml_register_image(const char * name, const void * src);
N      |             ^                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
_/user_src/examples_gen.c:86:45: error: too many arguments to function call, expected 2, have 3
;   86 |         lv_xml_register_image(NULL, "wink", wink);
9      |         ~~~~~~~~~~~~~~~~~~~~~               ^~~~
V/work/lvgl/src/others/xml/lv_xml.h:42:13: note: 'lv_xml_register_image' declared here
P   42 | lv_result_t lv_xml_register_image(const char * name, const void * src);
N      |             ^                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
�/user_src/examples_gen.c:88:9: error: call to undeclared function 'lv_xml_register_subject'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
T   88 |         lv_xml_register_subject(NULL, "room_1_temp", &subject_room_1_temp);
      |         ^
6 errors generated.
xmake[2]: *** [CMakeFiles/lved-runtime.dir/build.make:92: CMakeFiles/lved-runtime.dir/user_src/examples_gen.c.o] Error 1
-make[2]: *** Waiting for unfinished jobs....
�/user_src/widgets/dark_slider/dark_slider_gen.c:111:5: error: call to undeclared function 'lv_obj_set_flag'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
K  111 |     lv_obj_set_flag(obj, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS, true);
      |     ^
1 error generated.
�make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:167: CMakeFiles/lved-runtime.dir/user_src/widgets/dark_slider/dark_slider_gen.c.o] Error 1
Pmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lved-runtime.dir/all] Error 2
%make: *** [Makefile:91: all] Error 2
.emmake: error: 'make -j8' failed (returned 2)

Build failed: Make failed:
[... 30 lines cut ...]
E   85 |         lv_xml_register_image(NULL, "lvgl_logo", lvgl_logo);
C      |         ~~~~~~~~~~~~~~~~~~~~~                    ^~~~~~~~~
V/work/lvgl/src/others/xml/lv_xml.h:42:13: note: 'lv_xml_register_image' declared here
P   42 | lv_result_t lv_xml_register_image(const char * name, const void * src);
N      |             ^                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
_/user_src/examples_gen.c:86:45: error: too many arguments to function call, expected 2, have 3
;   86 |         lv_xml_register_image(NULL, "wink", wink);
9      |         ~~~~~~~~~~~~~~~~~~~~~               ^~~~
V/work/lvgl/src/others/xml/lv_xml.h:42:13: note: 'lv_xml_register_image' declared here
P   42 | lv_result_t lv_xml_register_image(const char * name, const void * src);
N      |             ^                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
�/user_src/examples_gen.c:88:9: error: call to undeclared function 'lv_xml_register_subject'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
T   88 |         lv_xml_register_subject(NULL, "room_1_temp", &subject_room_1_temp);
      |         ^
6 errors generated.
xmake[2]: *** [CMakeFiles/lved-runtime.dir/build.make:92: CMakeFiles/lved-runtime.dir/user_src/examples_gen.c.o] Error 1
-make[2]: *** Waiting for unfinished jobs....
�/user_src/widgets/dark_slider/dark_slider_gen.c:111:5: error: call to undeclared function 'lv_obj_set_flag'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
K  111 |     lv_obj_set_flag(obj, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS, true);
      |     ^
1 error generated.
�make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:167: CMakeFiles/lved-runtime.dir/user_src/widgets/dark_slider/dark_slider_gen.c.o] Error 1
Pmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lved-runtime.dir/all] Error 2
%make: *** [Makefile:91: all] Error 2
.emmake: error: 'make -j8' failed (returned 2)

mrhubbs, 2025-04-27:
Did you click the "generate code" button before compiling? These look like errors that result from omitting that step. Also, I presume you are using the example project from the latest master of this repo. From the logs, I can tell you are using Editor v0.2.0-rc1.

LuisMperezM, 2025-04-27:
I'm using v.0.1.0, and I'm doing the steps you gave me and the same thing keeps happening.

kisvegabor, 2025-05-01:
Hey,

We have just released v0.2, see [here](https://www.linkedin.com/posts/lvglgui_vscode-activity-7323482442684583937-i-Mk?utm_source=share&utm_medium=member_desktop&rcm=ACoAACUVOfsBKTiB9-maAgNttG_kJBYFAZ5NRrc).

Please try out with that.

closed 2025-10-20 by kisvegabor