lvgl/lvgl_pro #31: implementation issue(s) and preliminairy findings

h3lpd3sk, 2025-05-09:
First of all the 0.2 version looks really promissing!
A lot of hard work has gone in again. 
 
I'm having some troubles running the project on an actual device. 

Concider the following basic example (TI-RTOS).
On a memory contrainted MCU. 

Where lv_tick_inc is implemented in the lv_port_disp_init() with timer and functions.
This just renders a grey screen? 
Diving in deeper if rendering is done off canvas or something.

```
void uiTask(uintptr_t arg0, uintptr_t arg1) {

    lv_port_disp_init((SPI_Handle) arg0);
    lv_port_disp_wake();

    ui_init("S:/ui/examples/");
    lv_obj_t* home = home_create();

    const uint32_t timeout = 5;

    for (;;) {
        lv_timer_handler();

        Task_sleep(timeout);
    }
}
```
In my efforts to resolve the issue I found the following:

**1) compile errors**
if LV_USE_XML is set to 0 in lv_conf.h
There are compile errors in:
dark_slider_xml_parser.c
slider_box_xml_parser.c
easily fixed with #if LV_USE_XML  #endif
I will try to make a PR

**2) lvgl version**
I have lvgl as git submodule in the project, updated to last commit on master. 
But it is unclear to me what lvgl version/commit is used in the editor sandbox? 
So I think it is a good idea to denote somewhere in the info what lvgl version is required. 
And align it with a specific version of lvgl. 
Again I would prefer to the ability to use a local lvgl distribution to prevent sandboxing issues. 

**3) unknown define**
there is a condition for LV_EDITOR_PREVIEW which is not in master lv_conf.h

I can post my lv_conf.h if needed

h3lpd3sk, 2025-05-09:
I see I'm on an old version of the example. 
Let me update my findings :-)

h3lpd3sk, 2025-05-09:
For other people the way to initialize the ui screen is now
```
    ui_init("S:/ui/");
    lv_obj_t* main = main_create();
    lv_scr_load(main);
```

My previous points 1) and 3) no longer exist in the updated example

kisvegabor, 2025-05-13:
I'm happy the most of the issues are resolved . 😌 

> 2) lvgl version

It just a random commit as we add fixes/features to LVGL as we go. In the final version we will stick to released version. 
We are also thinking about allowing to use LVGL from a custom path. It'd be really powerful as you custom fork could be used as well. 

> 1) compile errors

@mrhubbs 
The generated `<widget>_xml_parser.c/h` should be guarded by `#if LV_USE_XML`.