lvgl/lvgl_pro #33: Slider Knob overshoots the endpoints by 1*Knob_Radius

FloHomi, 2025-05-13:
Hi There!
It appears that the knob of the slider is in the wrong position. (half it's radius). Is there a quick way to fix that? the problem is that a simple offeet in the one direction wouldn't do it. Then the knob is 2*R over on the other side.
It is hard to see, but I think in the online examples it is sligtly visible too.

[image]

kisvegabor, 2025-05-13:
Hey,

It's not related to the editor, it's more like a  general LVGL issue. 

Now the slider is assumed to be something like this: 

[image]

So the knob is always placed on the indicator. If it were inside the slider, the knob and indicator should go differently. It's also possible but not so trivial.

If the indicator is not visible you can do this:
```xml
<component>
    <styles>
        <!-- pad_hor moves the knob inwards-->
        <style name="main" bg_opa="255" pad_hor="14" height="25"/>
        
        <!-- Hide the indicator -->
        <style name="indic" bg_opa="0"/>
        
        <!-- pad_all < 0 make the knob smaller than the slider -->
        <style name="knob" bg_color="0xff0000" pad_all="-5"/>

        <!-- Remove the default knob grow animation on press -->
        <style name="knob_pr" transform_height="0" transform_width="0"/>
    </styles>

    <view width="100%">
        <lv_slider styles="main indic:indicator knob:knob knob_pr:knob:pressed"/>
    </view>
</component>
```

FloHomi, 2025-05-13:
Thank you for your quick answer 👍

kisvegabor, 2025-05-13:
It was the first time I shared editor code publicly in a issue, so it was my pleasure. 😊

Closing this issue.

closed 2025-05-13 by kisvegabor

FloHomi, 2025-05-15:
May I reopen this in LVGL Issues? Or can you move it?
I tried the solution from you, but It would look so much nicer, if the bar indicator would move too.

FloHomi, 2025-05-27:
I opend a new Issue: https://github.com/lvgl/lvgl/issues/8300
I noticed that the lv_arc is working as expected.