lvgl/lvgl_pro #38: FreeBSD / *BSD support.

cederom, 2025-05-20:
Hello world :-)

I am keeping fingers for LVGL Editor... but it is not available natively on FreeBSD / *BSD platforms as there is no source code available and Docker is Linux only solution (emulated on Windows and macOS).

Would it be possible to release the project so it can work natively on Open-Source platforms? :-)

Thanks :-)
Tomek

kisvegabor, 2025-05-22:
Hey Tomek,

Probably we will move to Podman from Docker. Do you know if it works on freeBSD?

cederom, 2025-05-23:
Wow, this podman seems to be working on FreeBSD, it would be really nice to try out with LVGL Editor, thank you @kisvegabor  :-)

https://wiki.freebsd.org/Containers

> podman, buildah
> 
> A FreeBSD port of the https://github.com/containers stack. Install sysutils/podman-suite from FreeBSD Ports/Packages.
> * Suitable for testing and evaluation.
> * Container storage using the zfs and vfs storage drivers. ZFS is strongly preferred since its use of snapshots and clones makes it more efficient than vfs.
> * Podman provides a CLI which is a drop-in replacement for docker.
> * Supports docker-style networking using a port of https://github.com/containernetworking/plugins.
> * Container images use the same formats and infrastructure as containerd and can be shared between the two implementations. 

Alternatively we can just create a native port for FreeBSD that would allow people to build it locally and/or install pkg package.. if the project is Open-Source.. if not then we could provide a precompiled binary package that should work as well :-)

closed 2025-10-20 by kisvegabor

cederom, 2025-10-20:
uhm okay :-(

EduardoMorras, 2025-10-28:
One more for BSD version. Thanks

mrhubbs, 2025-11-27:
Have any of you tried the LVGL Editor VS Code extension in VS Code on FreeBSD?

@EduardoMorras @nachtbraker @cederom