lvgl/lvgl_pro #40: Improve compatibility with embedded devices

drabart, 2025-05-22:
I tried to put the example onto a screen controlled by ESP32 and I encountered multiple issues:
 - There is no option to use simple lvgl include path (`#include "lvgl.h"` instead of `#include "lvgl/lvgl.h"`)
 - For some reason `lv_obj_set_flag` does not exist in lvgl 9.2.2 that ESP32 uses, so I had to replace it with `lv_obj_update_flag` (might just be lvgl versioning issue on my part)
 - The clock roller required `\n` instead of `&#xA` which was not replaced in generated code
 - The array generated for font did not have `const` specifier, which caused building issues due to insufficient RAM on ESP32
 - The clock label in the header was not bound to the subjects
 - The images did not show up. Probably due to issues with placing them in correct path, but I think it would be nice to be able to use the `lv_image_dsc_t` format for them, as they are much easier to use in that format on embedded devices
 - All the images are bell icon, because something about autogenerated passing of correct image sources does not work

Also as mentioned in the plan of the project the plugins will be for enterprise only, but is there any plan to allow for creating any device specific ports? Because I believe it would be great to be able to automatically generate lvgl and device specific files (component CMakeFiles.txt in case of ESP32) and I think it would not be feasible for the lvgl_editor team to maintain them for all possible devices.

Despite the issues I encountered, I really like the `developer-centric` approach you took.

kisvegabor, 2025-06-24:
Thank you for the extensive feedback! 

Most of the code gen issues will be fixed in v0.3 (planned to be released this week).

Regarding exporting images as arrays, see the options here: https://docs.lvgl.io/master/details/auxiliary-modules/xml/images.html

> The array generated for font did not have const specifier, which caused building issues due to insufficient RAM on ESP32

@mrhubbs please take a look at this.

> There is no option to use simple lvgl include path (#include "lvgl.h" instead of #include "lvgl/lvgl.h")

@mrhubbs 
We should have [this](https://github.com/lvgl/lvgl/blob/master/src/font/lv_font_montserrat_8.c#L7-L11) when `lvgl.h` is included. Later it would be even better to have a section like 
```xml
<code_export 
    lvgl_include_path="path/to/lvgl.h"
    other_options
 />
```
> Also as mentioned in the plan of the project the plugins will be for enterprise only, but is there any plan to allow for creating any device specific ports? Because I believe it would be great to be [...]

Generating a generic CMakeFile will the also part of v0.3. We are definitely interested in going deeper with chip vendor collaborations, be we can focus more on it when v0.1 is released in this October. 

> Despite the issues I encountered, I really like the developer-centric approach you took.

Thank you very much! Very very excited that how it is/will be received. 🤞

giobauermeister, 2026-01-22:
I think we could close this one since with the latest LVGL Pro versions we are heavily using the generated code in ESP-IDF environments and compiling with no issues. The CMakeLists.txt of the UI project should be customized accordingly. The idea is for it to be as generic as possible so it can be integrated in different environments.

Two examples here: 
https://github.com/giobauermeister/metronome_ui_lvgl_pro
https://github.com/giobauermeister/google_drive_menubar_lvgl_pro