lvgl/lvgl_pro #41: fatal error: lvgl/lvgl.h: No such file or directory

muyuballs, 2025-05-24:
compile use pio will see 
```
fatal error: lvgl/lvgl.h: No such file or directory
```

pls use
```c
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```
replace simple
```c
#include "lvgl/lvgl.h"
```
in gen header file

mkiepfer, 2025-06-04:
I ran into the same situation and I think the INCLUDE_SIMPLE flag may not help here, as the code generated by the LVGL editor doesn't recognize this flag. The generated code hard-coded uses "lvgl/lvgl.h"

muyuballs, 2025-06-05:
may in generated code not hard-coded 
```c
#include "lvgl/lvgl.h"
```
use a if-else micro
```c
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```

kisvegabor, 2025-06-07:
I see it now! Thank you! 
@mrhubbs Please use 
```c
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```

instead of simply including `lvgl/lvgl.h`