lvgl/lvgl_pro #44: Runtime file not found.

npwh, 2025-05-30:
I clicked 'clear cache' and then the preview window didn't show up.

The messages below are from the console:

Console Ready
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]

kisvegabor, 2025-05-30:
Have you also clicked the Build button?

npwh, 2025-05-30:
I’ve corrected the issue. It was caused by a component I created that had a problem, preventing the system from functioning. Once I deleted the problematic component and rebuilt, everything is back to normal.

closed 2025-05-30 by npwh