lvgl/lvgl_pro #45: LVGL Editor 0.2.3 – Preview fails to render

EqScript, 2025-06-03:
Attempting to render the preview in LVGL Editor 0.2.3 results in a failure. The console output shows the following error:

```
Console Ready  
Creating script element: runtime-script-canvas  
Appending script to document head  
Initialization already in progress, skipping  
Runtime file not found.  
Script load error: [object Event]  
Initialization failed: Error: Script load error: [object Event]
```

[image]

The editor runs inside Docker, which appears to be functioning normally.

System Info:

    OS: Linux Mint 22.1 (kernel 6.8.0)

    Docker: v28.2.2 (Engine - Community)

    Docker Compose: v2.36.2

    Cgroup: v2

    CPU: 12-core x86_64

    Memory: 14.8 GiB

Docker Status:

    Docker is running

    Containers: 2 (none active)

    Images: 3

Steps to Reproduce:

    Launch LVGL Editor 0.2.3

    Open or create a project

    Click on a widget or component file

Expected Result:
Preview should render a widget

Actual Result:
Initialization fails with a script loading error

mrhubbs, 2025-06-03:
Hello @EqScript, the Preview relies on a runtime that needs to be compiled. The steps are to press the generate code button, then the compile button, then try previewing an XML file.

Does that help?

EqScript, 2025-06-04:
Hi @mrhubbs — thanks for your input! You were absolutely right — a classic RTFM case on my end.

It didn’t compile at first due to some incorrect values being generated (e.g., 50% instead of plain uint8_t in icon_gen.c), but I patched that and it’s running smoothly now.

Thanks again!

— Sergio

closed 2025-06-04 by EqScript