lvgl/lvgl_pro #51: UI editor lags with larger project

jorticus, 2025-06-09:
I'm testing out a larger UI, and though it is not particularly complex, as I add more elements I have got to the point where updates to the UI take >1 second to complete, for *every change* I put into the editor.

This causes the text editor to freeze for up to a second for every keypress, which makes it very difficult to write new code.

As a workaround, it would be nice to have a way to disable auto-UI updates and make it manual refresh only.

However I'm not sure what's causing the perf issue - if you have any ideas of things I could collect to help pin it down, I'd be happy to help.

mrhubbs, 2025-06-11:
@jorticus Thanks for your valuable feedback! Can you share a sample project which causes this problem? If not, can you give an approximate idea of how many .xml files, how many images, etc. are in this larger UI?

I suspect that many of the files are being (unnecessarily) resent to the Preview when it refreshes.

kisvegabor, 2025-06-15:
I think it happens because of the fonts. Is it the same if you don't use custom fonts?

jorticus, 2025-06-15:
I uploaded my sample project here:

https://github.com/jorticus/lvgl_editor_mediaplayer

There are two images, and I am using a custom font  (TitilliumWeb-Light.ttf).

If I remove the custom font and switch back to the provided `Inter-SemiBold.ttf` then performance does improve a lot. So I guess the perf issue is due to the rendering of my specific font

As a side note, if I change the font in the globals.xml, the UI doesn't reflect it until I clear cache + regen code + full re-compile.

kisvegabor, 2025-06-24:
It's already fixed locally and will be part of v0.3 (targeting this week).

closed 2025-07-26 by mrhubbs