lvgl/lvgl_pro #54: no preview

weety, 2025-06-11:
I follow the step but no preview, why?

Console Ready
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Initialization already in progress, skipping
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]
File saved successfully

GageSt, 2025-06-14:
I opened the dev tools, here is the log when trying to generate a preview:
```
index-D4SgIjBF.js:181598 Creating script element: runtime-script-canvas
index-D4SgIjBF.js:181598 Appending script to document head
index-D4SgIjBF.js:181598 Initialization already in progress, skipping
index-D4SgIjBF.js:182484 
        
        
       GET app://runtime/lved-runtime.js net::ERR_ABORTED 404 (Not Found)
(anonymous) @ index-D4SgIjBF.js:182484
loadRuntimeScript @ index-D4SgIjBF.js:182448
initRuntime @ index-D4SgIjBF.js:182501
processData @ index-D4SgIjBF.js:182608
await in processData (async)
(anonymous) @ index-D4SgIjBF.js:182661
Qj @ index-D4SgIjBF.js:4954
Hk @ index-D4SgIjBF.js:6189
(anonymous) @ index-D4SgIjBF.js:6059
J2 @ index-D4SgIjBF.js:384
R2 @ index-D4SgIjBF.js:412
index-D4SgIjBF.js:181598 Script load error: [object Event]
index-D4SgIjBF.js:181598 Initialization failed: Error: Script load error: [object Event]
index-D4SgIjBF.js:182481 Uncaught (in promise) Error: Script load error: [object Event]
    at script.onerror (index-D4SgIjBF.js:182481:16)
script.onerror @ index-D4SgIjBF.js:182481
await in script.onerror (async)
(anonymous) @ index-D4SgIjBF.js:182661
Qj @ index-D4SgIjBF.js:4954
Hk @ index-D4SgIjBF.js:6189
(anonymous) @ index-D4SgIjBF.js:6059
J2 @ index-D4SgIjBF.js:384
R2 @ index-D4SgIjBF.js:412
index-D4SgIjBF.js:181598 Creating script element: runtime-script-canvas
index-D4SgIjBF.js:181598 Appending script to document head
index-D4SgIjBF.js:182484 
        
        
       GET app://runtime/lved-runtime.js net::ERR_ABORTED 404 (Not Found)
(anonymous) @ index-D4SgIjBF.js:182484
loadRuntimeScript @ index-D4SgIjBF.js:182448
initRuntime @ index-D4SgIjBF.js:182501
processData @ index-D4SgIjBF.js:182608
await in processData (async)
(anonymous) @ index-D4SgIjBF.js:182665
Qj @ index-D4SgIjBF.js:4954
Hk @ index-D4SgIjBF.js:6189
(anonymous) @ index-D4SgIjBF.js:6059
J2 @ index-D4SgIjBF.js:384
R2 @ index-D4SgIjBF.js:412
index-D4SgIjBF.js:181598 Script load error: [object Event]
index-D4SgIjBF.js:181598 Initialization failed: Error: Script load error: [object Event]
index-D4SgIjBF.js:182481 Uncaught (in promise) Error: Script load error: [object Event]
    at script.onerror (index-D4SgIjBF.js:182481:16)
script.onerror @ index-D4SgIjBF.js:182481
await in script.onerror (async)
(anonymous) @ index-D4SgIjBF.js:182665
Qj @ index-D4SgIjBF.js:4954
Hk @ index-D4SgIjBF.js:6189
(anonymous) @ index-D4SgIjBF.js:6059
J2 @ index-D4SgIjBF.js:384
R2 @ index-D4SgIjBF.js:412

```

kisvegabor, 2025-06-15:
Cc @mrhubbs

anujdeshpande, 2025-06-27:
+1

See the same on a Win11 machine - can confirm Docker is running