lvgl/lvgl_pro #63: Editor stuck

renzpDspread, 2025-06-18:
i use windows Editor, Docker Desktop.
Editor stuck encountered twice.
I waited for 5-10 minutes, but still couldn't work. Until I restart.

[image]

kisvegabor, 2025-06-24:
We have fixed an issue locally where a missing `<view>` tag caused freeze, but you have the <view> tag and it still freezes. 

Do you have any ideas how to reproduce it?

lonhtm, 2025-06-30:
Same to me. I noticed that the console always hang at ```Creating virtual filesystem``` like this when  the app was freezing

[image]

kisvegabor, 2025-07-10:
Please try with `v0.3.0-rc2`: https://github.com/lvgl/lvgl_editor/releases/tag/v0.3.0-rc2