lvgl/lvgl_pro #67: Docker required in Linux?

madduci, 2025-06-21:
Hello

I'm testing the application under Linux (Manjaro) and I've discovered over the console that it attempts to download and execute a docker image.

[image]

Even letting it downloading and running the image, the Project in this repository doesn't get rendered at all:

[image]

What am I missing?

madduci, 2025-06-24:
I have already configured it, but what has with the issue to do?

My question was more about why is Docker required and why isn't it
mentioned in the README?

Message ID: ***@***.***>
>

mrhubbs, 2025-06-24:
@madduci We use a Docker container as a predictable build environment, to compile LVGL. Once the image has downloaded, you need to click the "generate code" and then the "compile" buttons in the top right of the Preview panel, then you will see the Preview.

We don't like requiring Docker. It is heavy and a big install. We are working on a lightweight, built-in alternative for the next release.

madduci, 2025-06-30:
ok thanks for the reply.

closed 2025-06-30 by madduci