lvgl/lvgl_pro #72: Include path quirks using Arduino IDE

cjbaar, 2025-06-29:
Let me first say this looks like an amazing project. I can definitely see using it to streamline GUI creation, and I am very interested to see where it goes.

I am curious to know if anyone has tried to use this in an Arduino project (ESP32 based). I have struggled to get things compiled, and to get the Arduino IDE to properly locate and include all of the generated files. I am not sure how it is intended to be included. Here is what I have done to make it work.

1) LVGL Include
Arduino does not understand the path in the statement `#include "lvgl/lvgl.h"`. I am not sure what other environments use the directory format, but in Arduino IDE, libraries are loaded from a known path. Therefore, I have to change every statement (which is in multiple generated files) to just `#include "lvgl.h"`. I can do this with a post-processing script, but that is certainly cumbersome every time code is generated. I have `#define LV_LVGL_H_INCLUDE_SIMPLE` in my lv_conf.h, but the UI Editor is not aware of that file. I am not sure if there is a way to control this output from the XML.

2) Local Upstream Includes
The compiler in the Arduino IDE also does seem to find files upstream automatically. Specifically, I am referring to component/screen/etc files using `#include "ui.h"`. Arduino cannot seem to figure out how to include this from a parent directory. Instead, I have to change all of these references to use the relative path with `#include "../ui.h`. This happens regardless of project location (see #3).

3) Project Location
I first tried creating a project in the "libraries" folder, and including it as a library. I have seen this suggested online for other lvgl examples. In this case, if the library is called something like "lvgl_gui_test", then the library would be loaded by using `#include <lvgl_gui_test.h>`. However, when using this approach, the main Arduino sketch does not find functions in sub-folders. Therefore, I am unable to actually load any screen or components. For example, if I try to call `main_create()`, which is the generated function from `screens/main.xml`, the function is not found. Instead of using a library (as I have seen suggested online, the code seems to work if it is put inside the `src` directory of the Arduino sketch. In this case, the UI is initialized directly with `#include "src/lvgl_gui_test/ui.h"`.

I would be interested to know what experience others have had who have tried this. Overall, I am not sure how a project gets included with this or other development environments. Also, I find that when searching for things with "lvgl ui editor," I just get results that involve Squareline Studio.

kisvegabor, 2025-07-10:
We definitely need docs for Arduino.  If I understand correctly, something like this would be critical in  `project.xml`
```xml
<project>
  <code_gen lvgl_include_path="path/to/lvgl.h"/>
</project>
```
which replaces
```c
#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
    #include "lvgl.h"
#else
    #include "lvgl/lvgl.h"
#endif
```

with 
```c
#include "path/to/lvgl.h"
```

kisvegabor, 2025-10-20:
Unfortunately we didn't have time to add it in v1.0, butt it will be in v1.1. 🙁

laszlobalint, 2025-12-11:
@kisvegabor and @fbiego, do we need to act here?

fbiego, 2025-12-15:
@laszlobalint

The documentation that was recently added already explains some of the issues above and the available workarounds. That said, there still appears to be an export issue specific to widgets.

`widget_xml_parser.c` already handles the include correctly, but `widget_gen.c` does not currently account for this case and always exports:
```
#include "lvgl/src/core/lv_obj_class_private.h"
```

Perhaps we could also support something like
```
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
    #include "src/core/lv_obj_class_private.h"
#else
    #include "lvgl/src/core/lv_obj_class_private.h"
#endif
```

closed 2026-02-22 by mrhubbs