lvgl/lvgl_pro #74: LVGL Editor-0.3.0-rc (linux)

furiacdg, 2025-07-05:
Hi, I'm using LVGL for esp32 projects and LVGL Editor seem very good.

I tried to start new 0.3.0rc under linux (debian 11)

Here my feedback:

`Console Ready
Initiating container setup...
Checking for Container image...
Failed to list Container images
Failed to list container images
Container image update not required
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime script is loaded
Initializing runtime module instance...
Instance created and ready to use
Creating virtual filesystem
Initialization failed: Error: Aborted(Assertion failed: Cannot call unknown function lvrt_xml_load_translations, make sure it is exported)
`

`[Resource Converter]
Starting Container to convert 0 fonts and 3 images
/tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman)
/tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman)
/tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman)
/tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_LVGL Eh4AHBa/resources/internal-bin/podman)
Error: ❌ Error converting fonts/images
    at convertFontsAndImages (/tmp/.mount_LVGL Eh4AHBa/resources/app.asar/node_modules/@lved/shared-node/node_modules/@lved/utils/dist/generators/resource_converter.js:258:19)
    at async ProjectGenerator.generate (/tmp/.mount_LVGL Eh4AHBa/resources/app.asar/node_modules/@lved/shared-node/node_modules/@lved/utils/dist/generators/project_generator.js:469:9)
    at async ProjectService.generateProjectCode (/tmp/.mount_LVGL Eh4AHBa/resources/app.asar/node_modules/@lved/shared-node/dist/containers/project-management/components/services/project-service.js:606:13)
    at async generateAllCode (/tmp/.mount_LVGL Eh4AHBa/resources/app.asar/out/main/index.js:885:12)
    at async /tmp/.mount_LVGL Eh4AHBa/resources/app.asar/out/main/index.js:912:10
    at async Session.<anonymous> (node:electron/js2c/browser_init:2:106823)
❌ Error generating code: ❌ Error converting fonts/images`

TIA

kisvegabor, 2025-07-05:
Hi,

Please try clicking the "Clean" button first, then "Code export", and "Build". 

We need to streamline it if it helps.

furiacdg, 2025-07-06:
> Hi,
> 
> Please try clicking the "Clean" button first, then "Code export", and "Build".
> 
> We need to streamline it if it helps.

Hi, I've tried:

- Clean button: OK
- Generate code: I receive this error:

/tmp/.mount_LVGL EQpvIcP/resources/internal-bin/podman: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_LVGL EQpvIcP/resources/internal-bin/podman)
Error: ❌ Error converting fonts/images
    at convertFontsAndImages (/tmp/.mount_LVGL EQpvIcP/resources/app.asar/node_modules/@lved/shared-node/node_modules/@lved/utils/dist/generators/resource_converter.js:258:19)
    at async ProjectGenerator.generate (/tmp/.mount_LVGL EQpvIcP/resources/app.asar/node_modules/@lved/shared-node/node_modules/@lved/utils/dist/generators/project_generator.js:469:9)

mrhubbs, 2025-07-08:
Hi @furiacdg, this should be fixed in our release this week: v0.3.0-rc2.

After MUCH testing and investigation, we have determined the best path on Linux is for users to install `podman` with their package manager. More information will come in the release.

jakkra, 2025-07-09:
I'm having the same issue but when building. It's complaining about GLIBC_2.38
Do you have any minimum version required as I'm on ubuntu 22 the GLIBC version may be limited.

kisvegabor, 2025-07-10:
Note that on Linux with `v0.3.0-rc2` you need install podman with `sudo apt-get install podman`. It should resolve the issue. 

Closing this now. If it still doesn't work, please comment here and we will repoen it.

closed 2025-07-10 by kisvegabor