lvgl/lvgl_pro #77: lc_arc feature (or kinda help)

furiacdg, 2025-07-06:
Scope: create a gauge, a readonly arc with a percentage value displayed

My working example:

`
void lv_example_arc_1(void)
{
    lv_obj_t * label = lv_label_create(lv_screen_active());

    /*Create an Arc*/
    lv_obj_t * arc = lv_arc_create(lv_screen_active());
    lv_obj_set_size(arc, 150, 150);
    lv_arc_set_rotation(arc, 135);
    lv_arc_set_bg_angles(arc, 0, 270);
    lv_arc_set_value(arc, 50);
    lv_obj_center(arc);
    lv_obj_add_event_cb(arc, value_changed_event_cb, LV_EVENT_VALUE_CHANGED, label);
    
    // read only
    lv_obj_remove_style(arc, NULL, LV_PART_KNOB);
    lv_obj_remove_flag(arc, LV_OBJ_FLAG_CLICKABLE);

    /*Manually update the label for the first time*/
    lv_obj_send_event(arc, LV_EVENT_VALUE_CHANGED, NULL);
}

static void value_changed_event_cb(lv_event_t * e)
{
    lv_obj_t * arc = lv_event_get_target_obj(e);
    lv_obj_t * label = (lv_obj_t *)lv_event_get_user_data(e);

    lv_label_set_text_fmt(label, "%" LV_PRId32 "%%", lv_arc_get_value(arc));

    /*Rotate the label to the current position of the arc*/
    lv_arc_rotate_obj_to_angle(arc, label, 25);
}
`

My attempt:

    <view extends="lv_arc"
        bind_value="$position"
        styles="arc_thin"
        style_line_dash_width="1"
    >
        <lv_label align="center"  style_text_color="0xcccccc"></lv_label>
        <lv_event-call_function callback="myEventArc" trigger="value_changed" user_data="lv_label"/>
    </view>

Generated code:

`
    lv_obj_add_event_cb(lv_arc_1, myEventArc, LV_EVENT_VALUE_CHANGED, "lv_label");
`

It is possible to put in user_data something other than a string ?

Correct syntax should be:

`
    lv_obj_add_event_cb(lv_arc_1, myEventArc, LV_EVENT_VALUE_CHANGED,  lv_label_1);
`

Other question: it is possible to add in xml the "remove_style" and "remove_flag" syntax ?

Thanks

nurdism, 2025-07-06:
in rc 0.3 they have added a `<remove_style selector="knob" />`  tag, I'm also looking for a way to add `lv_obj_remove_flag`

kisvegabor, 2025-07-10:
> in rc 0.3 they have added a <remove_style selector="knob" /> tag

Thanks! Yes, it's really there and can be used. 🙂 

> I'm also looking for a way to add lv_obj_remove_flag

You can just set a flag to false. E.g. `<lv_obj scrollable="false"/>`

kisvegabor, 2025-07-28:
I forgot to answer on the user data question. It can be only string now. 

We might look into supporting other types as well, but unfortunately it's not trivial. 

As all the questions are answered I close this issue.

closed 2025-07-28 by kisvegabor