lvgl/lvgl_pro #93: color format error

sova32, 2025-07-16:
```
<globals>
    <api>
        <!-- Add <enumdefs> here -->
    </api>

    <consts>
        <!-- Add <px>, <int>, <color> etc here -->
    </consts>

    <styles>  
        <!-- Add <style> tags here -->
        <style 
        name="style_view_1"
        bg_color="#c9c9c9"
        />
    </styles>

    <images>
        <!-- Add <file> or <data> tags here -->
    </images>

    <fonts>
        <!-- Add <bin> , <tiny_ttf>, <freetype> tags here -->
    </fonts>
</globals>
```

```
⛭ Running make to build runtime for Preview
make: make -j8
[ 16%] Building C object CMakeFiles/lved-runtime.dir/user_src/gui1.c.o
[ 50%] Building C object CMakeFiles/lved-runtime.dir/user_src/ui.c.o
[ 50%] Building C object CMakeFiles/lved-runtime.dir/src/runtime.c.o
[ 66%] Building C object CMakeFiles/lved-runtime.dir/user_src/gui1_gen.c.o
[ 83%] Building C object CMakeFiles/lved-runtime.dir/src/getrandom_compat.c.o
/user_src/gui1_gen.c:60:23: warning: passing 'const lv_style_t *' to parameter of type 'lv_style_t *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]
   60 |         lv_style_init(&style_style_view_1);
      |                       ^~~~~~~~~~~~~~~~~~~
/work/lvgl/src/core/../misc/lv_style.h:345:33: note: passing argument to parameter 'style' here
  345 | void lv_style_init(lv_style_t * style);
      |                                 ^
/user_src/gui1_gen.c:61:52: error: expected expression
   61 |         lv_style_set_bg_color(&style_style_view_1, #c9c9c9);
      |                                                    ^
1 warning and 1 error generated.
make[2]: *** [CMakeFiles/lved-runtime.dir/build.make:92: CMakeFiles/lved-runtime.dir/user_src/gui1_gen.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lved-runtime.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
emmake: error: 'make -j8' failed (returned 2)
Build failed: Error: make failed when buiding the Preview runtime
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
```

mrhubbs, 2025-07-17:
@sova32 I'm sorry for the trouble. `#b73415` is an invalid color format, it has to be `0xb73415`. v0.3.0 of the Editor, being released very soon, will show you an error and highlight the erroneous color in the text editor.

kisvegabor, 2025-10-20:
Finally we decided to go for the `0x...` format only to avoid conflicts with `#constant`.

Closing this issue.