lvgl/lvgl_pro #97: "Generate code" does not create the Global Variables

walter-senekal, 2025-07-18:
Hi
I am using LVGL 0.3
I followed along with your youtube tutorial and made the changes as per the release notes

but when i click generate and compile , it fails with using of missing resources "temp1", "temp2", "temp3" and "hide"

when i look in "new_gen.c" and "new_gen.h" the "lv_subject_t temp1;" "lv_subject_t temp2;"  "lv_subject_t temp3;"  "lv_subject_t hide;" is missing

when i manually add then and there initializers and extern defs, it compiles perfectly.
But when i click "generate" again, it removes all references again

[new.zip](https://github.com/user-attachments/files/21318200/new.zip)

I have attached my project, and i also tried to append the worb "glob_" infront of each variable, just incase "hide" was a special keyword, but no change.

Best regards
Walter

mrhubbs, 2025-07-19:
@walter-senekal Thank you for reaching out to us. :) There issue is `<subjects>` appears twice in `globals.xml`. The first time it's blank. The second occurrence is being ignored.

We are planning to improve validation in the future so that duplicates of things that shouldn't be duplicated will show a clear error, instead of causing a silent problem.

Elias-EMCB, 2025-08-01:
Hi, I’m having the same problem. Do you have a solution? My project is ignoring the global variables. globals.xml