lvgl/lvgl_pro #102: Still No Preview on MacOS with v0.3

Liuweixian, 2025-07-25:
I can't get preview on my mac with v0.2 and v0.3
the console shows that

> Console Ready
> Checking Podman virtual machine status...
> Virtual machine is running
> Checking for Container image...
> Container Image found
> Creating script element: runtime-script-canvas
> Appending script to document head
> Initialization already in progress, skipping
> Runtime file not found.
> Script load error: [object Event]
> Initialization failed: Error: Script load error: [object Event]

Also I open the developer tools shows that 

Failed to load resource: the server responded with a status of 404 (Not Found)
index-J60fXg37.js:183882 Uncaught (in promise) Error: Script load error: [object Event]
    at script.onerror (index-J60fXg37.js:183882:16)
index-J60fXg37.js:183885 
            
            GET app://runtime/lved-runtime.js net::ERR_ABORTED 404 (Not Found)
(anonymous) @ index-J60fXg37.js:183885
loadRuntimeScript @ index-J60fXg37.js:183846
initRuntime @ index-J60fXg37.js:183902
processData @ index-J60fXg37.js:184011
index-J60fXg37.js:183882 Uncaught (in promise) Error: Script load error: [object Event]
    at script.onerror (index-J60fXg37.js:183882:16)
script.onerror @ index-J60fXg37.js:183882
script
(anonymous) @ index-J60fXg37.js:183849
loadRuntimeScript @ index-J60fXg37.js:183846
initRuntime @ index-J60fXg37.js:183902
processData @ index-J60fXg37.js:184011

How can I make this editor work ? Please help!

ManuelFeller, 2025-07-25:
Similar issue here. It either get stuck when saying
```
Compiling project
Checking Podman virtual machine status...
```
or shows the exact error from above. (I did not find out how to go back to that above state after clearing the cache though)

My Configuration:

- MacBook Pro with M1 Chip
- MacOS 15.5 (24F74)
- `which podman` returns `/opt/podman/bin/podman`
- podman Version is 5.5.0, the install directory it shows up in `$PATH`
- The `"podman" needs to be installed.` dialog shows up when opening LVGL UI Editor

podman is up and running according to the Podman Desktop UI, but I do not see any containers or pod being created when pressing the `Compile Project` button

Is there maybe any way I can start the container manually?

mrhubbs, 2025-07-25:
@ManuelFeller Thank you for all the details/troubleshooting. I think I know what is going on.

We actually do not support Podman from the official installer right now (although, of course, we should!). It is only supported when installed via Homebrew. That wouldn't be a problem since `podman` is in the `$PATH`... except that the `$PATH` on Mac for the terminal is often not the same as the `$PATH` used when launching a GUI application, like the Editor.

You have 3 options:

1. Uninstall Podman Desktop and install podman via brew.
2. Create a symlink on your machine like this: `/opt/homebrew/bin` -> `/opt/podman/bin`
3. Try launching the Editor from the terminal like this: `/Applications/LVGL\ Editor.app/Contents/MacOS/LV
GL\ Editor`. Possibly, that will be enough, or you may have to run `export PATH=$PATH:/opt/podman/bin` first.

Personally, I would try option 2.

I'm sorry for the inconvenience. In the next minor release, we will add `/opt/podman/bin` to the PATH the Editor uses when launching Podman, so it will support the officiaPodman installer.

ManuelFeller, 2025-07-25:
@mrhubbs I created the symlink with `ln -s /opt/podman/bin/podman /opt/homebrew/bin/podman` and now things work ( = I finally see the preview working as expected).

Thanks a lot for the fast reply!

Elias-EMCB, 2025-07-26:
"Hi, I'm a Windows user and the preview doesn't load in the desktop application. I'd like to know if it's something related to my configuration so I can fix it." 

[image]

Liuweixian, 2025-07-30:
@mrhubbs Thanks for your help, After clean cache, generate code and compile. I finally got preview showing in lvgl editor. But one more advise, I think this should be done at the first time of launching editor, like podman, So that I can get the ready to use environment for my development.

closed 2025-07-31 by mrhubbs