lvgl/lvgl_pro #106: lvgl editor error message

kokolina147, 2025-07-30:
✅ Built lib LVGL
⛭ Building runtime for Preview
configure: cmake -B /lved/project/build-preview -S /lved/resources -DPROJECT_NAME=lved-runtime -DOUTPUT_DIR=/lved/runtime -DLVGL_SRC_DIR=/lved/resources/lvgl -DLVGL_CONF_DIR=/lved/resources/conf -DUSER_SRC_DIR=/lved/project -DCMAKE_TOOLCHAIN_FILE=/emsdk/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake -DCMAKE_CROSSCOMPILING_EMULATOR=/emsdk/node/20.18.0_64bit/bin/node
-- The C compiler identification is Clang 21.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /emsdk/upstream/emscripten/emcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:40 (add_subdirectory):
  The source directory

    /lved/project

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:46 (target_compile_definitions):
  Cannot specify compile definitions for target "lib-ui" which is not built
  by this project.

CMake Error at CMakeLists.txt:61 (target_include_directories):
  Cannot specify include directories for target "lib-ui" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/lved/project/build-preview/CMakeFiles/CMakeOutput.log".
See also "/lved/project/build-preview/CMakeFiles/CMakeError.log".
emcmake: error: 'cmake -B /lved/project/build-preview -S /lved/resources -DPROJECT_NAME=lved-runtime -DOUTPUT_DIR=/lved/runtime -DLVGL_SRC_DIR=/lved/resources/lvgl -DLVGL_CONF_DIR=/lved/resources/conf -DUSER_SRC_DIR=/lved/project -DCMAKE_TOOLCHAIN_FILE=/emsdk/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake -DCMAKE_CROSSCOMPILING_EMULATOR=/emsdk/node/20.18.0_64bit/bin/node' failed (returned 1)
Build failed: Error: failed when buiding the Preview runtime
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
Creating script element: runtime-script-canvas
Appending script to document head
Initialization already in progress, skipping
Runtime file not found.
Script load error: [object Event]
Initialization failed: Error: Script load error: [object Event]"   i am getting this error after cloning the repository . i have been trying but have same issues

mrhubbs, 2025-08-07:
@kokolina147 Did that work for you?

kokolina147, 2025-10-07:
@mrhubbs Thnaks for that!