lvgl/lvgl_pro #130: problem with a property in a component placed in a widget

unitware, 2025-08-17:
I'm experiencing a problem with a simple button component in the making.
The property to set the text label is not included in the generated constructor

The component xml file is placed within a widget folder if it makes any difference

.. in deed it seems to be better if I move out the component to the components folder, manually delete the old generated files and finaly restart the editor.

1. I would like to be able to keep the widget specific components within the widget
2. Cut and paste files in the project explorer did copy but not delete the xml file.
3. I had to restart the editor to be able to use the component even after clearing cache files and rebuilding.

````
<component>
    <api>
        <prop name="a_text_property" type="string" default="+"/>
    </api>

    <view extends="lv_button"
         width="50"
         height="50"
         style_bg_color="#gray">

        <lv_label text="$a_text_property"
                  align="center"
                  style_text_align="center" 
                  >
      </lv_label>
   </view>
</component>
````

compile error:
````
...
/lved/project/widgets/digits/digit_button_gen.c:91:35: error: use of undeclared identifier 'a_text_property'

/lved/project/widgets/digits/digits_gen.c
:87:58: error: 
too many arguments to function call, expected single argument 'parent', have 2 arguments
   91 |     lv_label_set_text(lv_label_0, a_text_property);
      |                                   ^

   87 |     lv_obj_t
* digit_button_0 = digit_
button_create(obj, "+");
|                                 ~~~~~~~~~~~~~~~~~~~      ^~~
/lved/project/widgets/digits/digit_button_gen.h:43:12: 
note: 'digit_button_create' declared here
   43 | lv_obj_t * digit_button_create(lv_obj_t * parent);
      |            ^                   ~~~~~~~~~~~~~~~~~

1 error generated.
...
````

generated code (slightly shortened):
````
/** @file digit_button_gen.h */

lv_obj_t * digit_button_create(lv_obj_t * parent);
void digit_button_set_a_text_property(lv_obj_t * digit_button, const char * a_text_property);
````

````
static void digit_button_constructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
    LV_UNUSED(class_p);
    LV_TRACE_OBJ_CREATE("begin");

    digit_button_t * widget = (digit_button_t *)obj;
    
    static bool style_inited = false;

    if (!style_inited) {

        style_inited = true;
    }

    lv_obj_t * lv_button_0 = lv_button_create(obj);
    lv_obj_set_width(lv_button_0, 50);
    lv_obj_set_height(lv_button_0, 50);
    lv_obj_set_style_bg_color(lv_button_0, GRAY, 0);

    lv_obj_t * lv_label_0 = lv_label_create(lv_button_0);
    lv_label_set_text(lv_label_0, a_text_property);
    lv_obj_set_align(lv_label_0, LV_ALIGN_CENTER);
    lv_obj_set_style_text_align(lv_label_0, LV_TEXT_ALIGN_CENTER, 0);

    digit_button_constructor_hook(obj);

    LV_TRACE_OBJ_CREATE("finished");
}
````

````
static void digits_constructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
    LV_UNUSED(class_p);
    LV_TRACE_OBJ_CREATE("begin");

    digits_t * widget = (digits_t *)obj;
    
    static bool style_inited = false;

    if (!style_inited) {

        style_inited = true;
    }
    lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_COLUMN);
    lv_obj_set_style_pad_column(obj, 10, 0);

    lv_obj_t * digit_button_0 = digit_button_create(obj, "+");    // <<<-------

    digits_constructor_hook(obj);

    LV_TRACE_OBJ_CREATE("finished");
}
````

kisvegabor, 2025-08-26:
Thank you for reporting these

@mrhubbs I think these are mostly for you.

kisvegabor, 2025-10-20:
It should be already fixed in RC2. Closing this issue.

closed 2025-10-20 by kisvegabor