lvgl/lvgl_pro #132: problem with subject_increment_event in widget

unitware, 2025-08-17:
Am I doing something wrong here?

````
<widget>
   <api>
        <prop name="subject_" type="subject" />
        <prop name="digit_value" type="int" default="1" help="digit value {1, 10, 100}" />
   </api>

   <view flex_flow="column" style_pad_column="10">
      <digit_button a_text_property="+" >
         <subject_increment_event max="9" min="0" step="$digit_value" subject="$subject_" trigger="clicked" />
      </digit_button>
   </view>
</widget>
````

````
[ 20%] Building C object project/CMakeFiles/lib-ui.dir/widgets/digits/digits_xml_parser.c.o
cd /lved/project/build-preview/project && /emsdk/upstream/emscripten/emcc -DLV_EDITOR_PREVIEW=1 -DLV_USE_XML=1 @CMakeFiles/lib-ui.dir/includes_C.rsp -s USE_SDL=2 -std=gnu11 -MD -MT project/CMakeFiles/lib-ui.dir/widgets/dose_slider/dose_slider.c.o -MF CMakeFiles/lib-ui.dir/widgets/dose_slider/dose_slider.c.o.d -o CMakeFiles/lib-ui.dir/widgets/dose_slider/dose_slider.c.o -c /lved/project/widgets/dose_slider/dose_slider.c
cd /lved/project/build-preview/project && /emsdk/upstream/emscripten/emcc -DLV_EDITOR_PREVIEW=1 -DLV_USE_XML=1 @CMakeFiles/lib-ui.dir/includes_C.rsp -s USE_SDL=2 -std=gnu11 -MD -MT project/CMakeFiles/lib-ui.dir/widgets/digits/digits_xml_parser.c.o -MF CMakeFiles/lib-ui.dir/widgets/digits/digits_xml_parser.c.o.d -o CMakeFiles/lib-ui.dir/widgets/digits/digits_xml_parser.c.o -c /lved/project/widgets/digits/digits_xml_parser.c
/lved/project/widgets/digits/digits_gen.c
:88:56: error: use of undeclared identifier 'subject_'
   88 |     lv_obj_add_subject_increment_event(digit_butto
n_
0, subject_, LV_EVENT_CLICKED, digit_value, 0,
9);
      |                                                        ^

/lved/project/widgets/digits/digits_gen.c:88:84: 
error: use of undeclared identifier 'digit_value'
   88 |     lv_obj_a
d
d_subject_
increment_event(digit_button_0, subject_, LV_EVENT_CLICKED, digit_value, 0, 9);
      |                                                                                    ^

2
 errors generated.
make[2]: *** [project/CMakeFiles/lib-ui.dir/build.make:110: project/CMakeFiles/lib-ui.dir/widgets/digits/digits_gen.c.o] Error 1

make[2]: *** Waiting for unfinished jobs....

make[2]: Leaving directory '/lved/project/build-preview'
make[1]: *** [CMakeFiles/Makefile2:129: project/CMakeFiles/lib-ui.dir/all] Error 2
make[1]: Leaving directory '/lved/project/build-preview'
make: *** [Makefile:94: all] Error 2
make: Leaving directory '/lved/project/build-preview'
emmake: error: 'make -j 5 -C /lved/project/build-preview' failed (returned 2)
Build failed: Error: failed when buiding the Preview runtime
Failed to build the emscripten module
Failed to setup project: Error: Failed to build the emscripten module
````

kisvegabor, 2025-08-26:
For some reason the API property is not substituted. Let me csak it in my end. I believe it's already fixed.

kisvegabor, 2025-08-26:
I just realized what the problem is. In widgets you can't use $ to propagate properties but you need to write API functions.

@mrhubbs it causes many confusion so we need to validate the use of $ + API property names in widgets and give a validation error. 

Later we can also come up with s solution to support it.