lvgl/lvgl_pro #138: Convenience Feature Request:  Copy Path

vwheeler63, 2025-08-26:
If not already present or if not already planned, I would make regular use of a feature in the PROJECT FILES panel, in the right-click (context) menu of FOLDERS, to have a "Copy Path" option that would place the text path of that folder onto the clip board using the "/" as a directory separator.

- Copy Path

And so not as to assume what that text is going to be used for, perhaps the Windows version would offer an additional option:

- Copy Path (native directory separator)

kisvegabor, 2025-09-05:
Very good idea!

kisvegabor, 2025-09-27:
Added to our todo list.

laszlobalint, 2025-11-25:
@vwheeler63, copy path and copy relative path are now available context menu option in v1.0.1.