lvgl/lvgl_pro #145: 3D printed heater GUI - feedback

MitkoDyakov, 2025-09-01:
Hello all,

I am making a vat heater for my 3D printer. I am working with ESP32 S3, esp-idf, FreeRTOS and LVGL. First time user of all listed. I was pleased to find the LVGL UI Editor. 
You have done a lot of work and you have the right stuff in the project. However it has a long way to go. 

I want to share my observations in hopes it helps you. Some are small things, others I think are substantial, but they are not in any particular order.

1. When I open the editor it does not always shows me the start screen. There is no way to open a project from file menu. If I need to open two project at the same time, for example the project I am working on and the example project you provide, I need to open close the editor a couple if times. 
2. Names of components - some names of components are a trouble. For example I had one component name "target" and I got error in project.xml that I have not changed. Looking back I see my error but it took me some time to work it out. May be append automatically some suffix to user components name?
3. Figma + LVGL UI Editor - I liked the Figma UX editor. I think it is great and that you made good choice about selecting it. I paid for pro version and I got disappointed by the LVGL integration. I expected a XML definition of my project or someway of importing what i made into the LVGL Editor. All I got is pointing to components and getting the styles that I have typed in before hand... I feel cheated away from 24 Euro.
4. I need to format some strings for example: i want to print some values with leading 0s, 001, 002.... 012. This is usually done with %02i. But this does not work with your bind_text-fmt
5. For my project I have an indicator that should show the PWM value of a heater. It is a red dot, the harder the heater is driven the redder it should be. I wanted to bind a subject to the indicator opacity.  However I cannot bind integer subject to style opacity of element. 
6. Compatibility  - is there any way to select the LVGL version in the editor? I am running 9.3 on the ESP and I am having trouble with translations, binding styles a ton other things? I know it is not your issue, but you surely can fix it.
7. There is a problem with your example regarding "icon". Calling this function "lv_image_set_inner_align" before giving src to the icon breaks everything.
8. Passing subjects - why not directly pass the subject value to prop directly if type matches?  for example:Component "channel":
```
    <api>
        <prop name="power" type="int" default="255"></prop>
    </api> 
     ......
    <lv_label text="•" style_text_font="font_channel_dot" style_opa="$power" style_text_color="#red" x="9" y="9"> </lv_label>     
```
Screen:
  `  <channel channel="CH1" active="0" power="ch1_power" temp_small="ch1_temp_small" temp_big="ch1_temp_big" style_pad_all="0">`
Globals:
   `  <int name="ch2_power" value="0"/>`

This does not work. Why? But if I do the following:
    ` <channel channel="CH1" active="0" power="255" temp_small="ch1_temp_small" temp_big="ch1_temp_big" style_pad_all="0">`
Works just fine!

Hope this helps

You can find my project here:
https://github.com/MitkoDyakov/HeaterSoftware/tree/main/AI_GEN/GUI/HeaterGUI

Cheers,
Dimitar

kisvegabor, 2025-09-05:
> First I am using Windows. Here are the steps to reproduce. Open the editor and select a project. Leave this instance of the editor open. Start second instance and you will see the project that you have already opened and not the start page. To open a different project I close the second instance and start the editor for the third time and only then I can open a project different from the first.

It works well for me now, I hope t will be fixed for you too in the next release.

> When I say it does not work I mean in the editor. I cannot see the formatting in the Preview. Note: without the leading 0s it works fine. Here is how I use it:

Also works for me, hopefully next release too.

> I see. This brings a question what should be generated from the editor and what should be left to the user to code himself? I cannot give adequate response to this one. This remind me of STM32CubeIDE where the templates for project had space left for the user to populate with his code. The sections where marked by comments and after regeneration the cube would not touch whatever there was in the user section.

It's similar but without the comment section. Basically you can do in the editor what you can do in editor. 🙂 The rest, special code magic  can be added manually. 

> My controller can operate with two heaters: CH1 and CH2. I make component for a generic heater and use the api and prop to label it and show the temperature

If you just want to hide/show the dot can you
1. Create 2 subjects: `ch1_active`, `ch2_active` as you did
2. Bind the subjects to the hidden property of the labels by addig this a child `<bind_flag_if_eq ref_value="0" flag="hidden" subject="ch_active"/>`

> My question is why do I need to tell the api that the prop is a subject? Why this difference is made in the first place? For example:

When we process the XML "CH2" is a string. For example if you write `#CH2` or `$CH2` LVGL know that it needs to resolve the constant or API parameter to something else. But a string is just a string and will be used as it is. 

Anyways, I see the need for connecting subjects to local style properties.  E.g. 
```xml
  <lv_label>
       <bind_style_prop prop="bg_opa" subject="subject" selector="checked"/>
  </lv_label>
```

Just added it to idea list

> I found another thing in generating tab views. Here is the XML and the generated code.

What the problem here? Only the indentation?

kisvegabor, 2025-09-27:
Hi,

We are working on the project name change issue right now. 

Regarding `#T001` the editor things that it's a constant. We are figuring out something for that too.