lvgl/lvgl_pro #161: Failed to pull Container image

GregDuckworthRenishaw, 2025-09-24:
Hi,

I am testing LVGL Editor on my work PC to see if we could use it for an upcoming project, but I am running into issues after install.

I have installed wsl using wsl --install
I have installed LVGL editor 0.3.0
I then run LVGL editor and get the error:
`Console Ready
Preparing container environment...
Checking Podman virtual machine status...
Virtual machine is running
Checking for Container image...
Container Image not found
Pulling docker.io/lvglio/emscripten-sdl2:0.2.0... This process may take a few minutes.
Trying to pull docker.io/lvglio/emscripten-sdl2:0.2.0...
Error: unable to copy from source docker://lvglio/emscripten-sdl2:0.2.0: initializing source docker://lvglio/emscripten-sdl2:0.2.0: pinging container registry registry-1.docker.io: Get "https://registry-1.docker.io/v2/": tls: failed to verify certificate: x509: certificate signed by unknown authority
Failed to pull Container image: docker.io/lvglio/emscripten-sdl2:0.2.0`

Any ideas how I can fix the TLS error?

Thanks

Greg

kisvegabor, 2025-09-27:
cc @mrhubbs

laszlobalint, 2025-11-25:
@GregDuckworthRenishaw, can you, please, verify if the issue is resolved with the latest v1.01 release?
https://pro.lvgl.io/#download

closed 2025-12-01 by laszlobalint