lvgl/lvgl_pro #165: Invalid generated code when using "px" suffix

bhspyder, 2025-09-24:
When generating code from the following xml:
```xml
  <lv_button>
      <lv_label style_width="15px" text="Test 1" />
  </lv_button>
````
The generated code looks like this:
```c
lv_obj_set_style_width(lv_label_0, 15px, 0);
```

The additional px at the end of the 15 is not valid C/C++ code. The editor does not complain about the "px" being there (I believe it is valid xml).

Using % appears to correctly wrap the value in lv_pct.

bhspyder, 2025-10-06:
Just tested on RC2. Still an issue.

mrhubbs, 2025-10-06:
Removing the 'px' suffix was implemented for the types `int` and `px`, but not `coords`, which is what `width` is. Fixed. It'll be in the next release.