lvgl/lvgl_pro #170: Duplicate <subjects> elements will cause the project to fail to compile

lgl88911, 2025-09-27:
OS: Ubuntu 24.04
Editor: v0.3

When you two <subjects> in global.xml, the editor will ignore the second one, causing the compilation to fail. When I followed the video, I ignored the fact that there was already a <subjects> in global.xml and added another one myself, which caused the compilation to fail when it failed. This may be a small problem, but it is difficult to find the real problem from the error message. It would be perfect if it could indicate the problem in the XML.

```xml
    <subjects>
        <!-- Add <int>, <string>, or <float> subjects here -->
    </subjects>

    <styles>
        <!-- Add <style> tags here -->
    </styles>

    <subjects>
        <!-- Add <int>, <string>, or <float> subjects here -->
        <int name="focus" value="2"/>
        <int name="rest" value="1"/>
        <int name="sessions" value="1"/>
    </subjects>
```

error message：
```
>lved/project/components/setclock/setclock_gen.c:58:112: error: use of undeclared identifier 'focus'
   58 |     lv_roller_set_options(lv_roller_0, FOCUS_STRING, LV_ROLLER_MODE_NORMAL);lv_roller_bind_value(lv_roller_0, &focus);
      |                                                                                                                ^

/lved/project/components/setclock/setclock_gen.c:74:112: error: use of undeclared identifier 'rest'; did you mean 'rint'?
   74 |     lv_roller_set_options(lv_roller_1, BREAK_STRING, LV_ROLLER_MODE_NORMAL);lv_roller_bind_value(lv_roller_1, &rest);
      |                                                                                                                ^~~~
      |                                                                                                                rint

/emsdk/upstream/emscripten/cache/sysroot/include/math.h:319:13: note: 'rint' declared here
  319 | double      rint(double);
      |             ^

/lved/project/components/setclock/setclock_gen.c:90:115: error: use of undeclared identifier 'sessions'
   90 |     lv_roller_set_options(lv_roller_2, SESSION
S_STRING, LV_ROLLER_MODE_NORMAL);lv_roller_bind_value(lv_roller_2, &sessions);
      |                                                                                                                   ^

3 errors generated.
```

kisvegabor, 2025-09-28:
Hi,

Thank you for reporting it.

It's already fixed on our end and it will be part of the next release.

mrhubbs, 2025-10-07:
[image]

closing as fixed in v1.0.0-rc2

closed 2025-10-07 by mrhubbs